Industry4/MultiPlug.Ext.Hermes #43: Upstream SPI not receiving StopTransport

Industry4, 2022-09-30:
- PCB continues to move downline
- Connection disconnection occurs at inline with the Hermes Spec

[image]

majoon, 2022-10-06:
```
<Hermes Timestamp="2022-10-02T10:54:46.117">  
	<BoardAvailable BoardId="d2009cca-422f-11ed-8e19-88d7f6e3890a" BoardIdCreatedBy="PL0057" FailedBoard="1" ProductTypeId="" FlippedBoard="1" TopBarcode="0015225188" BottomBarcode="" Length="240" Width="110.152" Thickness="1" ConveyorSpeed="500" TopClearanceHeight="0" BottomClearanceHeight="0" Weight="0" /> 
</Hermes> 
02/10/2022 10:54:51 62 INFO SocketClient Sending: 
<?xml version="1.0" encoding="utf-16"?>
	<Hermes Timestamp="2022-10-02T10:54:51.809">
		<StartTransport BoardId="d2009cca-422f-11ed-8e19-88d7f6e3890a" ConveyorSpeed="500" />
	</Hermes>
02/10/2022 10:54:51 61 INFO SocketClient Data Sent
02/10/2022 10:54:53 60 INFO SocketClient Data Received: 
<Hermes Timestamp="2022-10-02T10:54:47.698">  
	<TransportFinished TransferState="3" BoardId="d2009cca-422f-11ed-8e19-88d7f6e3890a" /> 
</Hermes> 
02/10/2022 10:54:58 62 INFO SocketClient Sending: 
<?xml version="1.0" encoding="utf-16"?><Hermes Timestamp="2022-10-02T10:54:58.443">
	<CheckAlive Type="1" Id="cf8c153b-518a-4b74-b8fd-7fbcf7781a2c" />
</Hermes>
```

No StopTransport sent after TransportFinished.

Industry4, 2022-10-12:
This investigation highlighted a situation where SMEMA Machine Ready would not be toggled as expected, maybe due to the attached equipment being able to receive more than one PCB at a time. A new feature has been introduced **Stop Transport on Transport Finished** to cater for the scenario, bypassing the waiting of the SMEMA Machine Not Ready state.

### User Interface
**/extensions/multiplug.ext.hermes/lane/upstream/**
[image]

The Value of the Delay between **Stop Transport** sent and **Machine Ready** sent should be fairly long to take a good reading of the true SMEMA Machine Not Ready state. Setting this too short and a PCB may be transported into the attached equipment that wasn't intended. This feature will be available in **1.0.13**.

[image]

closed 2022-10-12 by Industry4