Industry4/MultiPlug.Ext.Hermes #45: Validate Scan Station Data

Industry4, 2022-09-30:
In order to prompt the user that the injected Scan Station data is bad, data inserted should be checked for known bad values.
For Example

- NOSCAN
- ERROR
- EMPTY
- NOT FOUND

These should be settable, and when to prompt the user should also be settable

- On Injection
- On Use

leeder269, 2022-10-03:
I would recommend to add a detection for double scan as well.

Industry4, 2022-10-06:
@leeder269  Thanks for your comment, this will be considered. You also might also want to consider setting the scan queue length to 1, this was added in version 1.0.12, more information here: #44

Industry4, 2023-04-15:
### Data Validation
/extensions/multiplug.ext.hermes/lane/scanstation/datasource/

This currently occurs **On Use** - ie When it is taken out of the Scan Station Queue
[image]

The Scan Station will show the **Verification** status as **Bad**. It has to be manually deleted.
[image]

closed 2023-04-16 by Industry4