Industry4/MultiPlug.Ext.Hermes #114: Duplicate SMEMA Machine Ready States after Saving

Industry4, 2024-06-28:
### Bug
Duplicate SMEMA Machine Ready States after Saving on the following pages:
- **_/extensions/multiplug.ext.hermes/lane/upstream/smema/_**
- **_/extensions/multiplug.ext.hermes/lane/upstream/smema/subscription/_**

### Screenshot
[image]

### Workaround

- This doesn't seem to effect runtime. All duplicate States change at the same time.
- The issue is resolved after software reboot

HaakonAxxe, 2024-08-02:
I have seen the samme issue at our site with duplicates.

Industry4, 2024-08-23:
This is available in 1.0.24 https://github.com/Industry4/MultiPlug.Ext.Hermes/releases/tag/1.0.24

closed 2024-08-23 by Industry4